aqeelanwar/PEDRA #3: np.reshape(0, 0, 3) error - from airsim

zvikarp, 2020-02-18:
hi again!

there is an error in airsim where it sometimes returns an empty image with length of 1. It would happed in the [get_depth()](https://github.com/aqeelanwar/PEDRA/blob/master/network/agent.py#L190) of [get_state()](https://github.com/aqeelanwar/PEDRA/blob/master/network/agent.py#L209) functions and they would return an error:

```
ValueError: cannot reshape array of size 1 into shape (0,0,3)
``` 

I have written a simple hacky workaround, do you want me to make a PR or give you the code here?

Do you also get it from time to time? It looks like there isn't a solution for it yet, see related issues on airsim repo:
https://github.com/microsoft/AirSim/issues/1840
https://github.com/microsoft/AirSim/issues/1710
https://github.com/microsoft/AirSim/issues/1755

annisanazi, 2020-02-21:
did you resolve this issue?

> hi again!
> 
> there is an error in airsim where it sometimes returns an empty image with length of 1. It would happed in the [get_depth()](https://github.com/aqeelanwar/PEDRA/blob/master/network/agent.py#L190) of [get_state()](https://github.com/aqeelanwar/PEDRA/blob/master/network/agent.py#L209) functions and they would return an error:
> 
> ```
> ValueError: cannot reshape array of size 1 into shape (0,0,3)
> ```
> 
> I have written a simple hacky workaround, do you want me to make a PR or give you the code here?
> 
> Do you also get it from time to time? It looks like there isn't a solution for it yet, see related issues on airsim repo:
> [microsoft/AirSim#1840](https://github.com/microsoft/AirSim/issues/1840)
> [microsoft/AirSim#1710](https://github.com/microsoft/AirSim/issues/1710)
> [microsoft/AirSim#1755](https://github.com/microsoft/AirSim/issues/1755)

did you resolve this issue?

aqeelanwar, 2020-02-21:
Hey,

I have never got this issue. But I have heard other people getting it.
One fix could be looking for the array returned. If its empty, may be the entire (s, a, s', r) can be dropped. This could work if the frequency of this error is not too much.

@zvikarp is this what you have done?

hshshs30202, 2020-03-14:
@zvikarp I have same error here. If you don't mind, Can i have your code solving this error?

zvikarp, 2020-03-16:
hi, the workaround i used only solves half of the problem.

in `agent.py` i have replaced this:
```
    def get_depth(self):
        responses = self.client.simGetImages([airsim.ImageRequest(2, airsim.ImageType.DepthVis, False, False)])
        depth = []
        img1d = np.fromstring(responses[0].image_data_uint8, dtype=np.uint8)
        depth = img1d.reshape(responses[0].height, responses[0].width, 3)[:, :, 0]
```

with this:
```
def img_to_1d(self, airsim_img):
        # securly convert the image to 1d array
        img1d = np.fromstring(airsim_img.image_data_uint8, dtype=np.uint8)
        if len(img1d) == 1:
            return img1d.reshape(1, 1, 1)
        return img1d.reshape(airsim_img.height, airsim_img.width, 3)

    def get_depth(self):
        responses = self.client.simGetImages([airsim.ImageRequest(2, airsim.ImageType.DepthVis, False, False)])
        depth = []
        depth = self.img_to_1d(responses[0])[:, :, 0]
```

 the error still sometimes occurs sometime in the `get_state()` function.

aqeelanwar, 2020-03-16:
PEDRA 2.0 addresses this problem by resetting the agent whenever this error occurs.

zvikarp, 2020-03-17:
thanks, you are awesome!

closed 2020-03-17 by zvikarp