aqeelanwar/PEDRA #7: Error when running main.py

jordon96, 2020-03-03:
when i tried to run main.py with a downloaded environment, i get this issue.

(airsim) C:\Users\Cagalli\Documents\Unreal Projects\MyProject3\Build\PEDRA>python main.py
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
------------------------------- configs/config.cfg -------------------------------
[general_params]
run_name:                      Tello_indoor
custom_load:                   True
custom_load_path:              models/trained/Indoor/indoor_cloud/Imagenet/e2e/e2e
env_type:                      Indoor
env_name:                      indoor_techno
phase:                         infer
SimMode:                       Multirotor
drone:                         DJIMavic
ClockSpeed:                    3
algorithm:                     DeepQLearning
ip_address:                    127.0.0.2
[camera_params]
width:                         640
height:                        360
fov_degrees:                   80
C:\Users\Cagalli\Documents\Unreal Projects\MyProject3\Build\PEDRA\util\transformations.py:1914: UserWarning: failed to import module _transformations
  warnings.warn('failed to import module %s' % name)
--------------------------- configs/DeepQLearning.cfg ---------------------------
[simulation_params]
load_data:                     False
load_data_path:                DeepNet/models/Tello_indoor/VanLeer/
[RL_params]
input_size:                    227
num_actions:                   25
train_type:                    e2e
wait_before_train:             10000
max_iters:                     150000
buffer_len:                    30000
batch_size:                    32
epsilon_saturation:            150000
crash_thresh:                  1.3
Q_clip:                        True
train_interval:                3
update_target_interval:        80000
gamma:                         0.99
dropout_rate:                  0.1
learning_rate:                 1e-06
switch_env_steps:              2000
epsilon_model:                 exponential
Traceback (most recent call last):
  File "main.py", line 13, in <module>
    eval(name)
  File "<string>", line 1, in <module>
  File "C:\Users\Cagalli\Documents\Unreal Projects\MyProject3\Build\PEDRA\algorithms\DeepQLearning.py", line 63, in DeepQLearning
    env_process, env_folder = start_environment(env_name=cfg.env_name)
  File "C:\Users\Cagalli\Documents\Unreal Projects\MyProject3\Build\PEDRA\aux_functions.py", line 40, in start_environment
    env_process = subprocess.Popen(path)
  File "C:\Users\Cagalli\Anaconda3\envs\airsim\lib\subprocess.py", line 707, in __init__
    restore_signals, start_new_session)
  File "C:\Users\Cagalli\Anaconda3\envs\airsim\lib\subprocess.py", line 992, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified

dinesh124, 2020-03-05:
I HAVE THE SAME ISSUE PLEASE RESPOND...

aqeelanwar, 2020-03-05:
Where did you save the environment that you downloaded?
It needs to be 

> PEDRA/unreal_envs/

closed 2020-04-14 by aqeelanwar