aqeelanwar/PEDRA #8: Error when running main.py

hshshs30202, 2020-03-11:
when running main.py
this error occured. I tried googling to find a solution, but i could not solve it.

 

C:\Users\user\DRLwithTL-master\util\transformations.py:1914: UserWarning: failed to import module _transformations
  warnings.warn('failed to import module %s' % name)
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html

Successfully loaded environment: indoor_techno
------------------------------ Drone ------------------------------
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

------------------------------ DQN ------------------------------

Loading weights from:  models/trained/Indoor/indoor_cloud/Imagenet/e2e/e2e
Fatal Python error: (pygame parachute) Segmentation Fault
TypeError: __init__() missing 3 required positional arguments: 'node_def', 'op', and 'message'

Thread 0x00000e40 (most recent call first):
  File "C:\Users\user\Anaconda3\envs\myenv\lib\threading.py", line 296 in wait
  File "C:\Users\user\Anaconda3\envs\myenv\lib\queue.py", line 170 in get
  File "C:\Users\user\Anaconda3\envs\myenv\lib\site-packages\tensorflow_core\python\summary\writer\event_file_writer.py", line 159 in run
  File "C:\Users\user\Anaconda3\envs\myenv\lib\threading.py", line 917 in _bootstrap_inner
  File "C:\Users\user\Anaconda3\envs\myenv\lib\threading.py", line 885 in _bootstrap

Thread 0x00002e94 (most recent call first):
  File "C:\Users\user\Anaconda3\envs\myenv\lib\threading.py", line 296 in wait
  File "C:\Users\user\Anaconda3\envs\myenv\lib\queue.py", line 170 in get
  File "C:\Users\user\Anaconda3\envs\myenv\lib\site-packages\tensorflow_core\python\summary\writer\event_file_writer.py", line 159 in run
  File "C:\Users\user\Anaconda3\envs\myenv\lib\threading.py", line 917 in _bootstrap_inner
  File "C:\Users\user\Anaconda3\envs\myenv\lib\threading.py", line 885 in _bootstrap

Current thread 0x00000620 (most recent call first):
  File "C:\Users\user\Anaconda3\envs\myenv\lib\site-packages\tensorflow_core\python\lib\io\file_io.py", line 384 in get_matching_files_v2
  File "C:\Users\user\Anaconda3\envs\myenv\lib\site-packages\tensorflow_core\python\lib\io\file_io.py", line 363 in get_matching_files
  File "C:\Users\user\Anaconda3\envs\myenv\lib\site-packages\tensorflow_core\python\training\checkpoint_management.py", line 366 in checkpoint_exists_internal
  File "C:\Users\user\Anaconda3\envs\myenv\lib\site-packages\tensorflow_core\python\training\saver.py", line 1280 in restore
  File "C:\Users\user\DRLwithTL-master\network\agent.py", line 458 in load_network
  File "C:\Users\user\DRLwithTL-master\network\agent.py", line 59 in __init__

zvikarp, 2020-03-16:
hey it looks like https://github.com/aqeelanwar/DRLwithTL is not supported any more, have you pulled the latest from this repository? 

can you please provide your `config.cfg` file and confirm that you added the env in `PEDRA/unreal_envs` as explined in https://github.com/aqeelanwar/PEDRA#createdownload-a-simulated-environment

hshshs30202, 2020-03-17:
@zvikarp Oh, That was my mistake.. thank you!

closed 2020-03-17 by hshshs30202