aqeelanwar/PEDRA #14: msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception.

hshshs30202, 2020-03-19:
I think that this issue is about num_agent settings. Could you help me to do this code..

[image]

---------------------------------- Environment ----------------------------------
Successfully loaded environment: indoor_techno

------------------------------------- Drone -------------------------------------
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "main.py", line 87, in <module>
    eval(name)
  File "<string>", line 1, in <module>
  File "C:\Users\hshsh\OneDrive\Desktop\pedra2\algorithms\DeepQLearning.py", line 26, in DeepQLearning
    client, old_posit, initZ = connect_drone(ip_address=cfg.ip_address, phase=cfg.mode, num_agents=cfg.num_agents)
  File "C:\Users\hshsh\OneDrive\Desktop\pedra2\aux_functions.py", line 247, in connect_drone
    client.enableApiControl(True, name_agent)
  File "C:\Users\hshsh\Anaconda3\envs\hs\lib\site-packages\airsim\client.py", line 37, in enableApiControl
    return self.client.call('enableApiControl', is_enabled, vehicle_name)
  File "C:\Users\hshsh\Anaconda3\envs\hs\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\hshsh\Anaconda3\envs\hs\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'drone0' is not available. This could either because this is simulation-only API or this vehicle does not exist.

aqeelanwar, 2020-03-19:
The issue could be the outdated environment. Can you download the environment from the link again and check if the error persists?

aqeelanwar, 2020-03-19:
Also, can you share the settings,json file? It should be in the documents folder.

hshshs30202, 2020-03-19:
It is updated environment version. Other environments doesn't work too. 
setting.json file is in C:\Users\hshsh\Documents\Airsim. Is this the right path?

aqeelanwar, 2020-03-19:
Did you update PEDRA to version 2.0 as well?
Yes that's the right path

hshshs30202, 2020-03-19:
Yes. I updated PEDRA version 2.0. Have you heard of any cases like this kind?

aqeelanwar, 2020-03-19:
I have not heard of any similar case.
If you could share config.cfg and settings.json file, I might be able to help

hshshs30202, 2020-03-19:
**config.cfg setting**
[general_params]
run_name:                   Tello_indoor
custom_load:                True
custom_load_path:           models/trained/Indoor/indoor_cloud/Imagenet/e2e/e2e
env_type:                   Indoor
env_name:                   indoor_long
mode:                       train
SimMode:                    Multirotor
drone:                      ARDrone
ClockSpeed:                 20
algorithm:                  DeepQLearning
ip_address:                 127.0.0.5
num_agents:                 1
[camera_params]
width:                      320
height:                     180
fov_degrees:                80

**setting.json setting**

{
    "Vehicles": {
        "drone0": {
            "Yaw": 180.0,
            "VehicleType": "SimpleFlight",
            "Z": 0,
            "X": 0.0,
            "Y": 0.0
        }
    },
    "ClockSpeed": 20,
    "CameraDefaults": {
        "CaptureSettings": [
            {
                "Height": 180,
                "ImageType": 0,
                "Width": 320,
                "FOV_Degrees": 80
            },
            {
                "Height": 180,
                "ImageType": 3,
                "Width": 320,
                "FOV_Degrees": 80
            }
        ]
    },
    "LocalHostIp": "127.0.0.5",
    "SettingsVersion": 1.2,
    "ViewMode": "NoDisplay",
    "PawnPaths": {
        "DefaultQuadrotor": {
            "PawnBP": " Class'/AirSim/Blueprints/BP_ARDrone.BP_ARDrone_C' "
        }
    },
    "SimMode": "Multirotor"
}

Thanks for you help..

hshshs30202, 2020-03-19:
OMG..There was another setting.json in C:\Users\hshsh\OneDrive\Documents\AirSim..
After moving setting.json file to this path, it works.
Sorry for taking your time thank U!

closed 2020-03-19 by hshshs30202

aqeelanwar, 2020-03-19:
Everytime you change the config.cfg file, settings.json is updated. 
I recommend you update the 'filename' variable in main.py on line 17 under the module generate_json to the path you moved settings.json to. This way you wont have to copy paste it everytime

`filename = 'C:\Users\hshsh\OneDrive\Documents\AirSim\settings.json'`