aqeelanwar/PEDRA #27: Error when running main.py

Lokesh97Bansal, 2020-04-18:
Dear Sir,
when I tried to run main.py, I get this issue.

(envPEDRA) E:\Major Project\PycharmProjects\PEDRA>python main.py
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
------------------------------- configs/config.cfg -------------------------------
[general_params]
run_name:                      Tello_indoor
env_type:                      Indoor
env_name:                      indoor_long
mode:                          infer
SimMode:                       Multirotor
drone:                         DJIMavic
ClockSpeed:                    20
algorithm:                     DeepQLearning
ip_address:                    127.0.0.5
[camera_params]
width:                         320
height:                        180
fov_degrees:                   80

---------------------------------- Environment ----------------------------------
Successfully loaded environment: indoor_long
E:\Major Project\PycharmProjects\PEDRA\util\transformations.py:1914: UserWarning: failed to import module _transformations
  warnings.warn('failed to import module %s' % name)
--------------------------- configs/DeepQLearning.cfg ---------------------------
[simulation_params]
custom_load:                   False
custom_load_path:              models/trained/Indoor/indoor_long/Imagenet/e2e/drone0/drone0
[RL_params]
input_size:                    103
num_actions:                   25
train_type:                    e2e
wait_before_train:             5000
max_iters:                     150000
buffer_len:                    10000
batch_size:                    32
epsilon_saturation:            100000
crash_thresh:                  1.3
Q_clip:                        True
train_interval:                2
update_target_interval:        8000
gamma:                         0.99
dropout_rate:                  0.1
learning_rate:                 2e-06
switch_env_steps:              2000000000
epsilon_model:                 exponential

------------------------------------- Drone -------------------------------------
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Initializing  DQN
Traceback (most recent call last):
  File "main.py", line 99, in <module>
    eval(name)
  File "<string>", line 1, in <module>
  File "E:\Major Project\PycharmProjects\PEDRA\algorithms\DeepQLearning.py", line 64, in DeepQLearning
    p_z,f_z, fig_z, ax_z, line_z, fig_nav, ax_nav, nav = initialize_infer(env_cfg=env_cfg, client=client, env_folder=env_folder)
  File "E:\Major Project\PycharmProjects\PEDRA\aux_functions.py", line 86, in initialize_infer
    f_z = env_cfg.floor_z/100
TypeError: unsupported operand type(s) for /: 'DotMap' and 'int'

aqeelanwar, 2020-04-18:
Can you share the following file?
unreal_envs/indoor_long/config.cfg

Lokesh97Bansal, 2020-04-18:
Dear Sir,
I noticed the file was missing at the specified location. Then I downloaded again the environment from the link
https://gtvault-my.sharepoint.com/personal/manwar8_gatech_edu/_layouts/15/onedrive.aspx?originalPath=aHR0cHM6Ly9ndHZhdWx0LW15LnNoYXJlcG9pbnQuY29tLzpmOi9nL3BlcnNvbmFsL21hbndhcjhfZ2F0ZWNoX2VkdS9FbkNjMnhXS0JDcElrV3NlZWdycDFFWUI2OXkwelMzbkxxenNhU2ZFLUtBRDRnP3J0aW1lPXllVXNpS1hqMTBn&id=%2Fpersonal%2Fmanwar8%5Fgatech%5Fedu%2FDocuments%2Fpublic%5Fshare%2FGitHub%2FPEDRA%2FPackaged%20Environments%2FIndoor%2Findoor%5Flong%2Erar&parent=%2Fpersonal%2Fmanwar8%5Fgatech%5Fedu%2FDocuments%2Fpublic%5Fshare%2FGitHub%2FPEDRA%2FPackaged%20Environments%2FIndoor

I am getting the simulation and the moving drone getting trained in it. But it is getting collided with the wall again and again and I have to start the simulation again. I am working on a project for autonomous landing of a drone on a moving platform, could you please guide me regarding how should I proceed further. I am quite new with Reinforcement Learning and autonomous systems. 

I am really thankful to you for taking your kind time to help me. 

Warm regards,
Lokesh

closed 2020-04-29 by Lokesh97Bansal