aqeelanwar/PEDRA #28: Value Error in Courtyard Map

nilesh-hegde, 2020-04-18:
I'm getting the following error in outdoor courtyard map

<class 'ValueError'> DeepQLearning.py 172
cannot reshape array of size 1 into shape (0,0)
Hit r and then backspace to start from this point

I've to reconnect and resume training sometimes

aqeelanwar, 2020-04-21:
At what iteration are you getting this error?

This issue is concerned with the generation of depth map. I have did a few changes to address this issue. Can you pull the latest repo and try if the error still persists?

nilesh-hegde, 2020-04-22:
This error is generated after about 2000 iterations or so.
I'll try with the latest repo once

nilesh-hegde, 2020-04-24:
I think the changes have worked as I have not come across it.
Thank you.

closed 2020-04-24 by nilesh-hegde