aqeelanwar/PEDRA #32: Linux support ?

oeeckhoutte, 2020-04-24:
Hello, 
I tried your project and downloaded everything (unreal and some environments). 
When I execute the main.py file I have the following error: 

OSError: [Errno 8] Exec format error: '...PEDRA/unreal_envs/indoor_long/indoor_long.exe' which make sense because I am on Ubuntu. Does this project support Ubuntu ? If yes could you help me ? Where can I find the compatible environments ?

UncleJo, 2020-07-29:
Even I'm trying to figure that out. 
The environments they have provided can run only on windows.
But they are prebuilt environments which runs on Unreal Engine and Airsim, which is only for windows. But the airsim APIs can be accessed over network so ubuntu should run fine, the only problem is that even tho the config.cfg lets you change the API access IP (default is 127.0.0.5 , ie local), the main.py file still tries to run the environment even tho it is running, on a different computer.

Is there way to not let PEDRA run the environment and call APIs via network? I'm unable to figure it out.

aqeelanwar, 2020-07-30:
@UncleJo 
You can edit the start_environment module by commenting the line 74 and un-commenting line 73
https://github.com/aqeelanwar/PEDRA/blob/3f2df7b68660c9d13651c16156ca9ce597d780d6/aux_functions.py#L69

This will disable the autostart of the environment and the code will look for any other already running environement. In this case you can start your environment before running main.py and then run main.py

aqeelanwar, 2020-07-30:
@oeeckhoutte 

I have not used PEDRA on Ubuntu OS. I can package the environment for Ubuntu, but I am not sure if it will work seamlessly with PEDRA or not. If you want, I can package one of the indoor environment for Ubuntu platform and share with you. May be you can experiment with it along with PEDRA.

Let me know.

UncleJo, 2020-07-31:
@aqeelanwar 
Thank you so much it started working after commenting the line.

aqeelanwar, 2020-08-02:
@UncleJo 
Did you create and package your own 3D environment for Ubuntu?

UncleJo, 2020-08-03:
Nope, I'm running Unreal Engine on my windows 10 PC along with Airsim and running the API in ubuntu 20.04.
Also I think the PEDRA is made for only Python 3.6.x as 3.8 is showing dependency error, some packages version (that is mentioned in requirements.txt) aren't available for Python 3.8.x.

aqeelanwar, 2020-08-04:
Yes. PEDRA requires python 3.6

Reuben-Isaac, 2020-08-16:
@aqeelanwar Would you be able to share an environment for Ubuntu? I would like to try experimenting with it. Thank you.

alirezadizaji, 2020-10-05:
Is there any way to export .uproject file from .exe file? if do so, then it could be used in Ubuntu by UE4Editor.

alirezadizaji, 2020-10-08:
> @oeeckhoutte
>
> I have not used PEDRA on Ubuntu OS. I can package the environment for Ubuntu, but I am not sure if it will work seamlessly with PEDRA or not. If you want, I can package one of the [...]

could you do that?, It would be fantastic if it supports Linux.

alirezadizaji, 2020-10-11:
Hi @aqeelanwar ,
could you share your binary output files from Unreal Engine, so that we could use it Ubuntu.
thanks in advance.

dwijaybane, 2020-10-16:
Hi @aqeelanwar please share 1 environment for ubuntu. Thank you.

danigr99727, 2021-02-26:
Hi @aqeelanwar could I also have at least one indoor environment for ubuntu? thanks!

COATZ, 2021-03-10:
Hi @aqeelanwar please share the forest outdoor env for Ubuntu. Thank you in advance!

ThorJonsson, 2021-04-13:
Hi, I would like to use your work for large scale experimentation on an academic cluster. Is this possible? Almost all academic clusters run Ubuntu or CentOS. If it is realistic to use these algorithms in a linux environment that would be amazing. If not do you know what are some potential alternatives one could consider for large scale experimentation?