aqeelanwar/PEDRA #47: Time Out Error

yeknafar, 2020-06-02:
Hello @aqeelanwar 

Thank you for your great project. 
I run this project in move_around mode without any error but when  I want to run this project in infer mode. when I try to run it I got this error:

"""
Traceback (most recent call last):
  File "main.py", line 104, in <module>
    eval(name)
  File "<string>", line 1, in <module>
  File "E:\Thesis\PEDRA\algorithms\DeepQLearning.py", line 28, in DeepQLearning
    client, old_posit, initZ = connect_drone(ip_address=cfg.ip_address, phase=cfg.mode, num_agents=cfg.num_agents)
  File "E:\Thesis\PEDRA\aux_functions.py", line 297, in connect_drone
    client.confirmConnection()
  File "C:\Users\hamed\anaconda3\envs\Thesis\lib\site-packages\airsim\client.py", line 54, in confirmConnection
    if self.ping():
  File "C:\Users\hamed\anaconda3\envs\Thesis\lib\site-packages\airsim\client.py", line 24, in ping
    return self.client.call('ping')
  File "C:\Users\hamed\anaconda3\envs\Thesis\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\hamed\anaconda3\envs\Thesis\lib\site-packages\msgpackrpc\future.py", line 43, in get
    raise self._error
msgpackrpc.error.TimeoutError: Request timed out

""""
can you help me please.
Thank you

aqeelanwar, 2020-06-04:
Can you please share your settings.json file from your documents folder?

yeknafar, 2020-06-04:
Thank you for your response.
This is my setting.json file

[](https://drive.google.com/file/d/1hu-1mYIR4_I2DB00ftNWpH-zi_iSj382/view?usp=sharing)

yeknafar, 2020-06-04:
I can't what is problem. but it solved without any change.
Thank you for your support.

aqeelanwar, 2020-06-04:
> I can't what is problem. but it solved without any change.
> but when I change mode in "train" or "infer" I can't see environment and quadcopter or environment not shown. what is problem ?
> 
> Thank you for your support.

By default the rendering is turned off to speed up the training procedure. You can hit the key '/' to render the environment and see the drone moving around.

yeknafar, 2020-06-04:
Thank you very much . :)))

aqeelanwar, 2020-06-04:
Did this solve your problem?

closed 2020-06-05 by yeknafar