aqeelanwar/PEDRA #48: tf.contrib.layers.flatten deprecated in tensorflow2

koonzi, 2020-06-03:
Hi, 

I am facing some issues while trying to make use of PEDRA. I am currently using tensorflow 2.0, and has used the tf_upgrade_v2 script to get PEDRA to run, however, it seems that tf.contrib.layers.flatten has been deprecated, and I've tried finding solutions to resolve it, and most of the solutions online suggested using tf.keras.layers.Flatten instead. I've tried using it and this is the error that I have faced, any help will be deeply appreciated!

------------------------------------- Drone -------------------------------------
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

----------------- drone0 -----------------
Initializing  DQN
WARNING:tensorflow:From C:\Users\teeko\venv\lib\site-packages\tensorflow\python\ops\resource_variable_ops.py:1666: calling BaseResourceVariable.__init__ (from tensorflow.python.ops.resource_variable_ops) with constraint is deprecated and will be removed in a future version.
Instructions for updating:
If using Keras pass *_constraint arguments to layers.
WARNING:tensorflow:From C:\Users\teeko\venv\lib\site-packages\tensorflow\python\ops\resource_variable_ops.py:1666: calling BaseResourceVariable.__init__ (from tensorflow.python.ops.resource_variable_ops) with constraint is deprecated and will be removed in a future version.
Instructions for updating:
If using Keras pass *_constraint arguments to layers.
Traceback (most recent call last):
  File "main.py", line 99, in <module>
    eval(name)
  File "<string>", line 1, in <module>
  File "C:\Users\teeko\ITPGitClone\PEDRA\algorithms\DeepQLearning.py", line 49, in DeepQLearning
    agent[name_agent] = PedraAgent(algorithm_cfg, client, name='DQN', vehicle_name=name_agent)
  File "C:\Users\teeko\ITPGitClone\PEDRA\network\agent.py", line 29, in __init__
    self.network_model = eval(net_mod)
  File "<string>", line 1, in <module>
  File "C:\Users\teeko\ITPGitClone\PEDRA\network\network_models.py", line 40, in __init__
    self.model = C3F2(self.X, cfg.num_actions, cfg.train_fc)
  File "C:\Users\teeko\ITPGitClone\PEDRA\network\network.py", line 39, in __init__
    self.flat = tf.keras.layers.Flatten(self.conv3)
  File "C:\Users\teeko\venv\lib\site-packages\tensorflow\python\keras\layers\core.py", line 631, in __init__
    self.data_format = conv_utils.normalize_data_format(data_format)
  File "C:\Users\teeko\venv\lib\site-packages\tensorflow\python\keras\utils\conv_utils.py", line 192, in normalize_data_format
    data_format = value.lower()
AttributeError: 'Tensor' object has no attribute 'lower'

yeknafar, 2020-06-05:
for flatten you can use this code in line 122 :
"
self.flat = tf.keras.layers.Flatten()(self.conv3)
"

after use this the problem of using flatten has been resolved but I got error using tf.nn.bias_add and I worked on it !

koonzi, 2020-06-06:
I am sorry for the trouble, but I have recently downgraded to an older version of TensorFlow, and your codes work fine! 

I am currently working on a project that is quite similar to PEDRA, and I would like to ask if I were to implement your codes on a real life drone for RL, what are some of the things I should take note of as your codes are implemented for AirSim?

aqeelanwar, 2020-07-30:
PEDRA focusses entirely on using AirSim for simulated 3D environment.
Implementing RL on a real drone has a lot of real-world issues such as

- You can't let the drone actually crash. You can use virtual crash, but that still has safety issues 
- You can't reset the drone position at the end of each episode (unless ofcourse a human interferes)
- RL is data hungry. You will run out of battery and will have to change multiple times.

You can look into the following paper that tries to address the issues mentioned above.
https://arxiv.org/abs/1807.08241

 May be that will give you a direction on implementing RL for real drone.

koonzi, 2020-07-30:
Alright, thank you for your help!

closed 2020-07-30 by koonzi

BazilaAfridi, 2021-12-12:
> for flatten you can use this code in line 122 : " self.flat = tf.keras.layers.Flatten()(self.conv3) "
> 
> after use this the problem of using flatten has been resolved but I got error using tf.nn.bias_add and I worked on it !

Can you please tell me that were you able to run PEDRA successfully with tensorflow version 2.x without downgrading to version 1?