aqeelanwar/PEDRA #51: CUDA dll import error

kfuller96, 2020-07-27:


aqeelanwar, 2020-07-30:
Hello,

Can you check if its a generic error running the following in python
`import tensorflow as tf`

I remember having issues regarding installing the correct CUDA version. I can check and share details on which version I used if you want.

kfuller96, 2020-07-30:
Hi Aqeel,
Run this in the virtual env or just in the general install?

aqeelanwar, 2020-07-30:
I meant execute in the python shell. In short, use the following commands

1. Open a terminal and activate your environment
``` conda activate myenv```
2. Start python shell
```python```
3. Type the following command and enter
```import tensorflow as tf```

kfuller96, 2020-07-30:
My apologies, Here is the output
[image]

aqeelanwar, 2020-07-30:
I use CUDA 9.0 on my GTX1080. May be try installing that.
Yes, the code will work on your GPU.

kfuller96, 2020-07-30:
Tried 9.0, same issue. But found this https://www.tensorflow.org/install/source_windows tutorial. Should I try and install tensorflow using these instructions and check if that fixes it?

aqeelanwar, 2020-08-02:
Yes. You should try that

kfuller96, 2020-08-02:
Alright, I had to install CUDA 8.0 and then cuDNN 6.0. that fixed everything.

closed 2020-08-03 by kfuller96

khlaifiabilel, 2022-05-18:
I have solved this particular problem by creating a Conda environment "python3.6".
Tested on both Windows 11 and Ubuntu 20 & 22