aqeelanwar/PEDRA #53: msgpackrpc.error

kfuller96, 2020-08-03:
Hi Aqeel,
I'm getting this error after a few hundred iterations. Hitting r and backspace doesn't seem to resolve it.
I've attached the setting.json file with this.

Cheers
[image]

[settings.txt](https://github.com/aqeelanwar/PEDRA/files/5013520/settings.txt)

aqeelanwar, 2020-08-04:
This usually happens when the connection between the Unreal Engine and AirSim is compromised. The python code tries to fetch the RGB frame or Depth map, and is not able to because of poor connectivity. I did take into account a few of the error catch to have a work around for this problem and it seemed to be working effectively.

How often do you get this error? also, can you share the results of the following print command?
https://github.com/aqeelanwar/PEDRA/blob/eb4e69aa3650ea4a96ee89104007ceecf7fb715f/algorithms/DeepQLearning.py#L451-L452

kfuller96, 2020-08-04:
It happens every time I run the sim,
Here is the output
[image]

aqeelanwar, 2020-08-05:
Incorporate the print commands in the python code. I have shared the link to the file above that will show you the placement of these commands in the DeepQLearning.py file.

Once you have incorporated the command, share the results of those print commands.

kfuller96, 2020-08-05:
My existing DeepQLearning.py file seems to have these print commands already present.
On a side note, do you mean run DeepQLearning.py by starting the simulation using "python main.py" which then fetches the DeepQLearning file?, Sorry, I'm relatively new to python so don't understand these things sometimes..

aqeelanwar, 2020-08-05:
No need to apologize.

Yes, try running the simulation through main.py. Those try and catch statement will print the error type as to what caused the problem. The error message will help us better understand the problem and how to solve it.

kfuller96, 2020-08-06:
Thanks Aqeel,
So I haven't had a repeat of the problem overnight, Simulation has passed 300k iterations with no issues. I suspect the problem could've been caused by my network card going to sleep!?. I wasn't able to change the power settings in windows so I ran a parallel steam download overnight to see if keeping the card in active use helped, which it has seemed to.
300k iterations and not error yet

Egoraptor09, 2020-08-07:
heres the time out error I got after 500k iterations and after incorporating the print command.

[image]

On a side note - Can the engine detect when the drone has successfully flown the env with crashing and save that model? Am i then able to load this pretrained model into another env? Lastly in inference mode, I'm assuming the control inputs are saved, how do I load them into "training mode"?

Thanks in advance, and for helping me out this far :)  🥇

aqeelanwar, 2020-08-18:
It depends on the nature of the algorithm you are trying to run. If you are running REINFORCE algorithm then every time the enviroment crashes you can load the previously saved model to begin the simulation again.

What do you mean by control inputs? The controls are output of the deep network in the inference mode. Based on what training learned, the deep network predicts the best output.

znkhn, 2022-11-01:
> This usually happens when the connection between the Unreal Engine and AirSim is compromised. The python code tries to fetch the RGB frame or Depth map, and is not able to because of poor [...]

Same error but ı have no 452 lines of code in my DeepQLearning.py file.