aqeelanwar/PEDRA #55: Getting real time output

Elnaz7393, 2020-08-17:
I need to use realtime images from the unreal environments and use them as input for my model to check if the current view is proper for moving forward or not and if it's not good my program calculates the angle and commands the drone to rotate. I want to do this at move_around mode.
How can I use `get_MonocularImageRGB()` to get image? I don't know what the arguments are for this function and what the output is. I don't have any idea what to do.
Please help me with what actions should I perform.

aqeelanwar, 2020-08-18:
The associated module can be found in aux_functions.py
https://github.com/aqeelanwar/PEDRA/blob/eb4e69aa3650ea4a96ee89104007ceecf7fb715f/aux_functions.py#L338

Elnaz7393, 2020-08-18:
I found it but what are `client` and `vehicle_name`?

Elnaz7393, 2020-08-19:
Hi 
I tried to get `client` value(I'm not sure if I said it correct) from `client, old_posit, initZ = connect_drone(ip_address=cfg.ip_address, phase=cfg.mode, num_agents=cfg.num_agents)` but this error occured
[image]

aqeelanwar, 2020-09-10:
Can you please share teh content of your config.cfg file?

Elnaz7393, 2020-09-10:
Thanks for your incredible code. 
The problem is solved by changing `mode` in config.cfg from `move_around` to `infer`.

closed 2020-09-10 by aqeelanwar