aqeelanwar/PEDRA #56: Unable to run main.py

kevantjh, 2020-08-18:
Hi sir, i'm getting this error when attempting to run the main.py file. Hoping u can help me out with this. 
[image]

aqeelanwar, 2020-08-18:
Hello,
It seems that the system wasn't able to find the downloaded/extracted unreal environment. Are you sure you downloaded the environment and extracted in the right place? The following link will help
https://github.com/aqeelanwar/PEDRA#download-a-simulated-environment

kevantjh, 2020-08-19:
Hi Sir,
Thanks i was able to get the program to run, however i can't seem to navigate the drone using my keyboard in move_around mode. Other functions like camera movement are working fine though. I'm new to python and running simulations so i greatly appreciate your help!

aqeelanwar, 2020-09-10:
Can you please share the content of your config.cfg file?

rakehsaleem, 2022-04-21:
Hi @kevantjh, were you able to run the main.py file in train or move_around mode successfully? I am having an issue running the main.py file and seeking any help. The move_around mode I can't control using a drone and also, while I train, it stuck at a point saying you have multiple names for variable "vehicles_names" which I printed and comes out to be 1. My other thread is #106 where I posted the complete issue!

Any response would be much appreciated!

LIXUANLIANG, 2022-05-31:
me too 
and my config.cfg file was 

[general_params]
run_name:                   Tello_indoor
env_type:                   Indoor
env_name:                   indoor_long
ip_address:                 127.0.0.1
algorithm:                  DeepQLearning
mode:                       train

[drone_params]
SimMode:                    Multirotor
drone:                      DJIMavic
ClockSpeed:                 20

[camera_params]
width:                      320
height:                     180
fov_degrees:                80

w4646, 2025-04-01:
> Hi Sir, Thanks i was able to get the program to run, however i can't seem to navigate the drone using my keyboard in move_around mode. Other functions like camera movement are working fine though. I'm new to python and running simulations so i greatly appreciate your help!

Hello，How do you solve this problem?