aqeelanwar/PEDRA #57: msgpackrpc.error

XueZhihan, 2020-08-24:
Hi sir, I'm getting this error when running the main.py.
```
Traceback (most recent call last):
  File "C:/SC/AirSim SC/PEDRA/main.py", line 99, in <module>
    eval(name)
  File "<string>", line 1, in <module>
  File "C:\SC\AirSim SC\PEDRA\algorithms\DeepQLearning.py", line 21, in DeepQLearning
    client, old_posit, initZ = connect_drone(ip_address=cfg.ip_address, phase=cfg.mode, num_agents=cfg.num_agents)
  File "C:\SC\AirSim SC\PEDRA\aux_functions.py", line 300, in connect_drone
    client.enableApiControl(True, name_agent)
  File "C:\tools\IDEs\Anaconda3\lib\site-packages\airsim\client.py", line 40, in enableApiControl
    return self.client.call('enableApiControl', is_enabled, vehicle_name)
  File "C:\tools\IDEs\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\tools\IDEs\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'drone0' is not available. This could either because this is simulation-only API or this vehicle does not exist.

```
Hoping u can help me out with this.

aqeelanwar, 2020-09-10:
Hello,
Can you please share the content of your config.cfg file?

bilalkabas, 2021-02-06:
Hi,
I was getting the same error since my Documents folder is inside my OneDrive folder. In `main.py`, it looks for `setting.json` in `C:\Users\user_name\Documents\Airsim\settings.json` directory. 

`path = os.path.expanduser('~\Documents\Airsim')`

So, I changed the above line in `main.py` like this

`path = os.path.expanduser('~\OneDrive\Documents\Airsim')`

Now, it is working.

levicorpus0407, 2021-02-20:
> Hi sir, I'm getting this error when running the main.py.
>
> ```
> Traceback (most recent call last):
>   File "C:/SC/AirSim SC/PEDRA/main.py", line 99, in <module>
>     eval(name)
>   File "<string>", line [...]
Hi, I encountered the same error... Did you solve the issue?
config.cfg:
[general_params]
run_name:                   Tello_indoor
env_type:                   Indoor
env_name:                   indoor_techno
ip_address:                 127.0.0.5
algorithm:                  DeepREINFORCE
mode:                       train

[drone_params]
SimMode:                    Multirotor
num_agents:                 1
drone:                      DJIMavic
ClockSpeed:                 20

[camera_params]
width:                      320
height:                     180
fov_degrees:                80

error info:
Traceback (most recent call last):
  File "F:/PEDRA/main.py", line 104, in <module>
    eval(name)
  File "<string>", line 1, in <module>
  File "F:\PEDRA\algorithms\DeepREINFORCE.py", line 27, in DeepREINFORCE
    client=client)
  File "F:\PEDRA\aux_functions.py", line 365, in connect_drone
    client.enableApiControl(True, name_agent)
  File "D:\Anaconda3\envs\python36\lib\site-packages\airsim\client.py", line 37, in enableApiControl
    return self.client.call('enableApiControl', is_enabled, vehicle_name)
  File "C:\Users\xzx\AppData\Roaming\Python\Python36\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\xzx\AppData\Roaming\Python\Python36\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'drone0' is not available. This could either because this is simulation-only API or this vehicle does not exist.

levicorpus0407, 2021-02-20:
> Hi,
> I was getting the same error since my Documents folder is inside my OneDrive folder. In `main.py`, it looks for `setting.json` in `C:\Users\user_name\Documents\Airsim\settings.json` directory. [...]

hi, it seems like my path setting is correct. still got the same error info

bilalkabas, 2021-02-21:
> hi, it seems like my path setting is correct. still got the same error info

I recommend you to locate the `settings.json`  file which is by default inside `Documents\AirSim` directory. Normally, after running the `main.py`, it should look like this

```json
{
    "SettingsVersion": 1.2,
    "LocalHostIp": "127.0.0.5",
    "SimMode": "Multirotor",
    "ClockSpeed": 20,
    "ViewMode": "FlyWithMe",
    "PawnPaths": {
        "DefaultQuadrotor": {
            "PawnBP": " Class'/AirSim/Blueprints/BP_DJIMavic.BP_DJIMavic_C' "
        }
    },
    "Vehicles": {
        "drone0": {
            "VehicleType": "SimpleFlight",
            "X": 0.0,
            "Y": 0.0,
            "Z": 0,
            "Yaw": 0.0
        }
    },
    "CameraDefaults": {
        "CaptureSettings": [
            {
                "ImageType": 0,
                "Width": 320,
                "Height": 180,
                "FOV_Degrees": 80
            },
            {
                "ImageType": 3,
                "Width": 320,
                "Height": 180,
                "FOV_Degrees": 80
            }
        ]
    }
}
```
If it is not like this for your case, you can copy above lines and paste them into your `settings.json` file and try again. This may help.

levicorpus0407, 2021-02-22:
> I recommend you to locate the `settings.json` file which is by default inside `Documents\AirSim` directory. Normally, after running the `main.py`, it should look like this
>
> ```json
> { [...]

Thank you for your help!
I copied your 'settings.json' into mine and ran the main.py. However, after function 'enableApiControl' (called with 2 arg(s)) threw an exception, the file has been rewrite to this automatically:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2
}
or else the other kind of error is msgpackrpc.error.TimeoutError: Request timed out
I checked out the generation_json(cfg) function in main.py separately, it worked just fine.
Then I tried to delete the parameter 'name_agent' in enableApiControl function as the error info told me... After removed many 'name_agent' parameter in the whole project, it seemed to run the algorithm without exception thrown... But I doubt that it can be debug this way...because the settings.json still isn't correct.

levicorpus0407, 2021-02-23:
Hi, I found that the error occurred due to my asian characters in settings.json file path. My user name is asian character, which may not supported by AirSim. After changing that, simulation began in the end.