aqeelanwar/PEDRA #58: The drone hits the wall and stuck at training in the loop in the env of indoor_vanleer. Could you help to modify the env or the drone settings?

BenoitKAO, 2020-09-01:
Thank you for your marvelous works!

I found the issue that the drone hits the wall and stuck at training in the loop in the env of indoor_vanleer.

Could you help to modify the env or the drone settings?

Here is the link of recording video:
https://youtu.be/E9hx_QkMAPE

Any help is appreciated.

aqeelanwar, 2020-09-03:
Hello,

Did you change the environment initial position? The default values were supposed to work fine.
Yes, you can modify the orientation of the drone at its initial position in the following module
https://github.com/aqeelanwar/PEDRA/blob/eb4e69aa3650ea4a96ee89104007ceecf7fb715f/unreal_envs/initial_positions.py#L33

The third argument is the yaw angle, you can change that to change the yaw of the drone in its initial position.

BenoitKAO, 2020-09-04:
Hi Aqeelanwar,

No, I did not change the environment initial position.

Following by your page:
https://github.com/aqeelanwar/PEDRA/blob/master/unreal_envs/readme.md

I do not know how to execute 'initial_positions.py'?
= = =
def indoor_vanleer():
    # The environment can be downloaded from
    # https://drive.google.com/drive/u/2/folders/1u5teth6l4JW2IXAkZAg1CbDGR6zE-v6Z
    orig_ip = [
        [-3100, -4530, 90],  #Player Start
        [-1340, -2240, -90],
        [-3790, -5450, 180],
        [-3980, -1760, -90]
    ]
    level_name = ['VanLeer1', 'VanLeer2', 'VanLeer3', 'VanLeer4']
    crash_threshold = 0.07
    return orig_ip, level_name, crash_threshold
= = =

aqeelanwar, 2020-09-10:
You can provide as many intial positions for the drone as you like. The algorithm alternates between the initial positions every 'switch_env_steps' as defined in the algorithm config file.
https://github.com/aqeelanwar/PEDRA/blob/eb4e69aa3650ea4a96ee89104007ceecf7fb715f/configs/DeepREINFORCE.cfg#L14

level_name are the names assigned to each of these initial positions. This could be anything that helps you understand which initial position the drone is at.

In order to select a specific initial position for the drone among the ones defined in the initial_position.py file, use the reset_to_inital

kirk0306, 2020-10-04:
Great work!
Also had similar issue but the drones only get stucked at ceiling (4 drones in indoor_complex), so I guess the problem is about initial height in Airsim. Any help is appreciated thnx.

aqeelanwar, 2020-10-09:
@kirk0306 Yes. That would be the issue. Did you modify the initial height of the drone?

kirk0306, 2020-10-09:
> @kirk0306 Yes. That would be the issue. Did you modify the initial height of the drone?

I did not modify the height, it seems fine after I add some delay when looping to `reset_to_initial` in DeepQlearning.py , and `connect_drone` in aux_functions.py . So I guess it just need time to let airsim respawn the agents one by one.

aqeelanwar, 2020-11-19:
@kirk0306 
Yes. You are right. I experimented with the delay as well. The drone needs time to settle to the defined height before it can be used again. I have edited the PEDRA code to address this issue. Thanks.

closed 2020-11-19 by aqeelanwar