aqeelanwar/PEDRA #59: checkpoint, weight, and custom_load_path, are confusing while performing the inference. Can you provide the tutorial document in detail?

BenoitKAO, 2020-09-03:
Thank you for your marvelous works!

I am confused at checkpoint, weight, and custom_load_path in inferencing,
and I cannot find the tutorial document in detail.

This is the setting of config.cfg:
===
[general_params]
run_name:                   Tello_indoor
env_type:                   Indoor
env_name:                   indoor_cloud
ip_address:                 127.0.0.5
algorithm:                  DeepREINFORCE
mode:                       infer

[drone_params]
SimMode:                    Multirotor
num_agents:                 1
drone:                      DJIMavic
ClockSpeed:                 100

[camera_params]
width:                      320
height:                     240
fov_degrees:                94
===

This is the setting of DeepREINFORCE.cfg:
===
[simulation_params]
custom_load:                True
custom_load_path:           models/trained/Indoor/indoor_cloud/Imagenet/e2e/drone0
distributed_algo:           GlobalLearningGlobalUpdate-SA

[RL_params]
input_size:                 103
num_actions:                25
train_type:                 e2e
total_episodes:             15000000
batch_size:                 32
crash_thresh:               1.3
learning_rate:              1e-4
switch_env_steps:           2000000000
gamma:                      0.99

[distributed_RL params]
communication_interval:     100
average_connectivity:       2
===

This is the content of checkpoint file:
===
model_checkpoint_path: "drone0_15300"
all_model_checkpoint_paths: "drone0_14900"
all_model_checkpoint_paths: "drone0_15000"
all_model_checkpoint_paths: "drone0_15100"
all_model_checkpoint_paths: "drone0_15200"
all_model_checkpoint_paths: "drone0_15300"
===

The unreal engine is crashed one time with uncertainty.
The env is restarted and reconnected.

Then, I have paused two times of training to save the weight, and then recover the connection to the unreal engine.
Continue the training.

I do not know why there are many checkpoints?

This is the screenshot of the folder:
https://drive.google.com/file/d/1novWNoKip4CqZdd7SDjkhvKn991vA4iv/view?usp=sharing

After the training processing is interrupted:
https://drive.google.com/file/d/1RrvFIriWNHuTOZwKRLMQP4va8Rup1ayJ/view?usp=sharing

I set 'custom_load_path:' to:
custom_load_path: models/trained/Indoor/indoor_cloud/Imagenet/e2e/drone0

The error messages were shown:
===
Fatal Python error: (pygame parachute) Segmentation Fault
TypeError: __init__() missing 3 required positional arguments: 'node_def', 'op', and 'message'

Thread 0x00006558 (most recent call first):
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\threading.py", line 295 in wait
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\queue.py", line 164 in get
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\site-packages\tensorflow_core\python\summary\writer\event_file_writer.py", line 159 in run
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\threading.py", line 916 in _bootstrap_inner
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\threading.py", line 884 in _bootstrap

Thread 0x000068d4 (most recent call first):
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\threading.py", line 295 in wait
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\queue.py", line 164 in get
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\site-packages\tensorflow_core\python\summary\writer\event_file_writer.py", line 159 in run
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\threading.py", line 916 in _bootstrap_inner
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\threading.py", line 884 in _bootstrap

Current thread 0x000065d8 (most recent call first):
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\site-packages\tensorflow_core\python\client\session.py", line 1443 in _call_tf_sessionrun
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\site-packages\tensorflow_core\python\client\session.py", line 1350 in _run_fn
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\site-packages\tensorflow_core\python\client\session.py", line 1365 in _do_call
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\site-packages\tensorflow_core\python\client\session.py", line 1359 in _do_run
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\site-packages\tensorflow_core\python\client\session.py", line 1180 in _run
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\site-packages\tensorflow_core\python\client\session.py", line 956 in run
  File "D:\Users\BenoitKAO\miniconda3\envs\p36\lib\site-packages\tensorflow_core\python\training\saver.py", line 1290 in restore
  File "D:\gggggggggggithhubss\P_S_v0\network\network_models.py", line 361 in load_network
  File "D:\gggggggggggithhubss\P_S_v0\network\network_models.py", line 244 in __init__
  File "<string>", line 1 in <module>
  File "D:\gggggggggggithhubss\P_S_v0\network\agent.py", line 29 in __init__
  File "D:\gggggggggggithhubss\P_S_v0\algorithms\DeepREINFORCE.py", line 58 in DeepREINFORCE
  File "<string>", line 1 in <module>
  File "main.py", line 99 in <module>
===

While I set 'custom_load_path:' to:
custom_load_path: models/trained/Indoor/indoor_cloud/Imagenet/e2e/drone0/drone0_14900
or
custom_load_path: models/trained/Indoor/indoor_cloud/Imagenet/e2e/drone0/drone0_15000
or
...
or
custom_load_path: models/trained/Indoor/indoor_cloud/Imagenet/e2e/drone0/drone0_15300

The drone is nonstop flying in a circle:
https://drive.google.com/file/d/12610RW8uNrlfgC2Xu_s6bYTNkJL0Q68L/view?usp=sharing

Can you provide the tutorial document in detail on how to correctly train/ infer, save the weight, set custom_load_path, ..., and so on?

Any help is appreciated!

aqeelanwar, 2020-10-09:
I apologize for the delayed response. Were you able to resolve the issue?

BenoitKAO, 2020-11-06:
Dear Aqeelanwar,

No,
this situation has occurred every time.

Thank you very much.

aqeelanwar, 2020-11-19:
For the custom_load_path, the correct way of initializing it is
`custom_load_path: models/trained/Indoor/indoor_cloud/Imagenet/e2e/drone0/drone0_xxxxx`
where drone0_xxxxx is the prefix of saved weights.

As for drone flying in circle, it is possible that your weights were not trained properly. Did you see increase in safe flight/return graoh on tensorboard when you were training. Additionally, you can check the output of your network in the infer mode to see if it always predicts the same action as output (may be move right). This can be the reason of the drone flying in a perfect circle.