aqeelanwar/PEDRA #64: No floor plan

motahare96, 2020-10-02:
Hi
Thanks for sharing your code. Unfortunately there's a problem with indoor long floor plan as I can't download it. It's necessary for infer mode.
with regards

aqeelanwar, 2020-10-07:
It should be in the zip file of the environment you downloaded by the name of long_floor.PNG.
Here is the link in case it was misplaced.
https://gtvault-my.sharepoint.com/:u:/g/personal/manwar8_gatech_edu/EYZQqeOIqsFEtaZOGzhzu6oBndlCiAAvDjTwbuh9n3R7jw?e=5VvVRs

siddy2000, 2020-10-09:
HI Mr. Aqeelan, your completed file of all of the environments seems to be expired.

aqeelanwar, 2020-10-09:
Hello,

I have updated the link. Let me know if you can access it now.

siddy2000, 2020-10-09:
Yes that works great. Thanks for everything! Your work is amazing.

motahare96, 2020-10-14:
Thanks for your response but .png file can't be downloaded.

aqeelanwar, 2020-11-19:
@motahare96 
Were you able to solve the issue?

siddy2000, 2020-11-19:
Yes I was. Thank you.