aqeelanwar/PEDRA #70: Unable to run main.py(D-PEDRA)

JaysonZhang1997, 2020-12-09:
Hi sir, i'm getting this error when attempting to run the main.py file. It stopped and couldn't open UE file. Hoping u can help me out with this.
[image]

JustinLeFever, 2020-12-10:
I have the same exact issue. This happens with other environments as well. I have put the indoor_complex and indoor_pyramid enivronments into the unreal_envs folder and changed the config accordingly, but this exact thing happens. Nothing actually launches, unreal engine doesn't even open. However, the single drone branch works just fine (apart from some package issues in the requirements text files that I had to work around).

Also, trying to run 'pip install requirements_gpu.txt' or 'pip install requirements_cpu.txt' in the multi drone version yields the error: 
ERROR: Could not find a version that satisfies the requirement requirements_gpu.txt
ERROR: No matching distribution found for requirements_gpu.txt

This doesn't happen in the single drone version. Not sure if its a big deal for me anyways though since I already installed the packages from the single drone version, but it might be a noteworthy bug.

aqeelanwar, 2020-12-10:
@JaysonZhang1997 
The issue has been resolved. The problem was in the start_environment module in the aux_functions.py. 
`# env_process = []
    env_process = subprocess.Popen(path)`
You can either pull from the repo, or fix the issue by changing the lines to the above ones.

Let me know if the issue still persists.

aqeelanwar, 2020-12-10:
@JustinLeFever 
Regarding the requirements.txt, did you try 
`pip install -r requirements.txt`

JustinLeFever, 2020-12-11:
@aqeelanwar 
Aha I somehow managed to forget the '-r' lol. Thanks, that worked for me. And thank you for getting back to me, very much appreciated.

aqeelanwar, 2020-12-14:
@JustinLeFever 
No problem.

@JaysonZhang1997 
Were you able to resolve the issue?

JaysonZhang1997, 2020-12-15:
> @JustinLeFever
> No problem.
> 
> @JaysonZhang1997
> Were you able to resolve the issue?

[image]
I was too busy some time ago. I just tried it and it didn't work.I deleted settings.json and tried again, but still failed. Thank you for your patience.

JaysonZhang1997, 2020-12-15:
Hi sir, it works on train/DeepQLearning mode.

> @JustinLeFever
> No problem.
> 
> @JaysonZhang1997
> Were you able to resolve the issue?

Hi sir, it works on train/DeepQLearning mode.