aqeelanwar/PEDRA #73: unable to see screens during simulation

edsun123, 2020-12-12:
I do not see the simulation screen with the drone. This includes the depth map and the segmentation map.

However, I can see the floorplan and the front facing camera.

aqeelanwar, 2020-12-12:
Hello,

Did you try using the key 1, 2, 3?
You can find more information here
https://github.com/aqeelanwar/PEDRA/blob/master/unreal_envs/readme.md#environment-supported-features

edsun123, 2020-12-13:
Yes, I have tried keys 1,2,3 because 3 was able to produce the front-facign camera. 

I think this is an issue with airsim.

aqeelanwar, 2020-12-13:
Can you please share the screenshot?

edsun123, 2020-12-13:
Here is screen shot in train mode: 
[image]

aqeelanwar, 2020-12-13:
Try using the key 'F'. This will enable the cameras and you will see the big black screen replaced with the feed from the front facing camera. Then try using keys 1 and 2.

edsun123, 2020-12-13:
Thank you! This helped.

 Is the key 'F' from the airsim instructions? 

(i can only find a reference of it https://github.com/microsoft/AirSim/blob/master/docs/using_car.md)

aqeelanwar, 2020-12-14:
Yes it is from AirSim. The 'F' key enables the graphics rendering. When it is enabled, only then you can view the depth and segment map in the simualtion. This rendering however doesn't affect the backedn python API. Python can still 'see' these images. Its just that these are not beging rendered for the users to see in the simualtion.

Graphics rendering for unreal simulation can take up significant compute power. So, I intentionally disable it when generating the settings.json file (No Display). This helps during the training phase, and each training iteration is much faster (about x2 times on GTX 1080) when the rendering is disabled. This means total training time is reduced by almost half.

edsun123, 2020-12-14:
Great! Thank you for the explanation. Hopefully this will be helpful for others. 

I will be closing this issue.

closed 2020-12-14 by edsun123