aqeelanwar/PEDRA #92: Missing Path Specified

BazilaAfridi, 2021-09-07:
Where can I find the models/trained/Indoor/indoor_long/Imagenet/e2e/drone0/drone0 path. There is no such path nor any models folder exist in PEDRA-PEDRA project  folder

jamilaandimen, 2021-09-07:
Hi, I don't think you need it if you leave `custom_load : False` in the config file like https://github.com/aqeelanwar/PEDRA/blob/master/configs/DeepQLearning.cfg for example

BazilaAfridi, 2021-09-07:
Ok thanks. Can anyone tell me why this error is occuring?

Traceback (most recent call last):
  File "C:\Users\Microsoft\Downloads\PEDRA-PEDRA\PEDRA-PEDRA\main.py", line 93, in <module>
    env_process, env_folder = start_environment(env_name=cfg.env_name)
  File "C:\Users\Microsoft\Downloads\PEDRA-PEDRA\PEDRA-PEDRA\aux_functions.py", line 73, in start_environment
    env_process = subprocess.Popen(path)
  File "C:\Users\Microsoft\anaconda3\envs\myvenv\lib\subprocess.py", line 947, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Microsoft\anaconda3\envs\myvenv\lib\subprocess.py", line 1416, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

jamilaandimen, 2021-09-08:
Did you download the simulated environment and set up everything like mentioned in read.md file before executing your code?

BazilaAfridi, 2021-09-08:
Yes I did activated the virtual environment, installed all the packages, downloaded and placed the outdoor_forest environment in the unreal_envs folder, edited the config file(also changed the env_name to outdoor_forest) and when run the python main.py command got this error.

BazilaAfridi, 2021-09-09:
I have found the solution by using another environment because there is no .exe file for the outdoor_forest environment.

jamilaandimen, 2021-09-09:
Thant's great :)