aqeelanwar/PEDRA #106: Error while running simulation in training mode --> msgpackrpc.error

rakehsaleem, 2022-04-13:
Hi @aqeelanwar, I'm trying to use this library to train and test the basic functionality of PEDRA but I'm running into some issues. At the moment, my config settings are like this:
```
[general_params]
run_name:                 Tello_indoor
env_type:                   outdoor
env_name:                 utdoor_oldtown
ip_address:                127.0.0.5
algorithm:                  DeepQLearning
mode:                         train

[drone_params]
SimMode:                   ComputerVision
num_agents:               1
drone:                         DJIPhantom
ClockSpeed:                20

[camera_params]
width:                          320
height:                        180
fov_degrees:               80
```

and I'm trying to train, while I run the command python main.py, the simulation starts but right after that I see the following error:

> msgpackrpc.error.RPCError: rpclib: client error C0002: Function 'simGetImages' was called with an invalid number of arguments. Expected: 3, got: 2

although a similar issue to this was posted to AirSim repository and I check that and tried to make changes but still, couldn't get it working. Any help on this would be much appreciated, thank you!

rakehsaleem, 2022-04-13:
**_Post Error:_**
I tried to add an argument '0' which I read in the AirSim repository so the function:

```
responses1 = client.simGetImages( 
    [
         airsim.ImageRequest('front_center', airsim.ImageType.Scene, False, False)
    ], vehicle_name=vehicle_name)  # scene vision image in uncompressed RGBA array
```
changed to 
```
responses1 = client.simGetImages(0, 
    [
         airsim.ImageRequest('front_center', airsim.ImageType.Scene, False, False)
    ], vehicle_name=vehicle_name)  # scene vision image in uncompressed RGBA array
```

after this, the simulation gives me the following error.

>   File "main.py", line 105, in <module>
>     eval(name)
>   File "<string>", line 1, in <module>
>   File "C:\PEDRA\algorithms\DeepQLearning.py", line 72, in DeepQLearning
>     current_state[name_agent] = agent[name_agent].get_state()
>   File "C:\PEDRA\network\agent.py", line 129, in get_state
>     camera_image = get_MonocularImageRGB(self.client, self.vehicle_name)
>   File "C:\PEDRA\aux_functions.py", line 404, in get_MonocularImageRGB
>     ], vehicle_name=vehicle_name)  # scene vision image in uncompressed RGBA array
> TypeError: simGetImages() got multiple values for argument 'vehicle_name'

I'm trying to debug the code and fix the problem but I'll update if I get success and if somebody figured it out, please drop the suggestion here. Thanks again!

rakehsaleem, 2022-04-13:
@aqeelanwar Still struggling with this error, and can not get it working. Any help or potential lead would be much appreciated. Thank you!

closed 2022-04-22 by rakehsaleem

xiaoshengjianan, 2023-01-07:
Can you tell me how to solve this problem?