aqeelanwar/PEDRA #115: How to fix destination?

zstar1003, 2022-08-27:
Hi, Sir. 
I read two papers mentioned in the repository and note that the indicator of the evaluation model is safe flight distance. According to my understanding, this work is to explore the longest time for UAV to avoid obstacles in a limited environment. 

I want to explore the shortest path and shortest time of UAV at two points. How can I fix destination and guide the UAV? Can you give me some tips? Thanks a lot.

chenyi10, 2023-08-26:
do you have any idea?

chenyi10, 2023-08-26:
hi,sir.
i have the same question,do you have any idea now

> Hi, Sir. I read two papers mentioned in the repository and note that the indicator of the evaluation model is safe flight distance. According to my understanding, this work is to explore the longest [...]