aqeelanwar/PEDRA #124: Unreal Engine shut down itself

Unsophisticate1, 2023-04-24:
When I run main.py, the environment can be opened automatically. However, after running about 100 episodes, the unreal engine shuts down itself and the terminal print messages like the following picture.How can I solve this problem?
[image]

Bruce-Si, 2023-05-08:
Hello, where did you download the environments since the link is down? tks.

Unsophisticate1, 2023-05-08:
You can download from here[image]
/

cyyispig, 2023-10-18:
Have you solved this problem?I also encountered this problem

Bruce-Si, 2023-10-20:
Sorry, I haven't solve it.

FerdinandHu99, 2024-02-20:
> 运行了一分钟就出现这个问题，有人解决了吗？

Lowering the Unreal Engine version to 4.18.3 solved it!